adafruit/Adafruit-Pi-Finder #19: Connection Refused

golddave, 2015-01-25:
I powered up my new Raspberry Pi with a bootable SD. I can see it on the network. When I run finder.sh it correctly identifies the IP address but when it attempts to connect via SSH it gets a connection refused message.

(I should note that this is my first Raspberry PI so it's possible that I'm missing something.)

toddtreece, 2015-01-25:
@golddave what OS are you using? When you say you are running finder.sh, are you running it via `curl -SLs https://apt.adafruit.com/bootstrap | bash`, or did you clone the repo?

toddtreece, 2015-01-25:
@golddave also, what version of raspbian did you install on your sd card?

brennen, 2015-01-26:
@golddave - my first guess on this one would be that the SSH server daemon just isn't enabled on your Pi. There's a bit of [a tutorial for enabling it](https://learn.adafruit.com/adafruits-raspberry-pi-lesson-6-using-ssh/enabling-ssh).  Apologies if you've already covered this base, but it's the first thing to eliminate.

I will look at amending finder.sh to tell the user if SSH doesn't appear to be available.

golddave, 2015-01-26:
I ended up connecting to a monitor. The SD card I had booted directly to NOOB.

closed 2015-01-26 by golddave

toddtreece, 2015-01-26:
@golddave ah. that makes sense. i updated the docs to reflect the raspbian SD card requirement. thanks for testing it out, and sorry it wasn't clear!  you can run `curl -SLs https://apt.adafruit.com/install | sudo bash` directly from the pi to run the bootstrap (just make sure you are plugged in to the network)

@brennen SSH is enabled by default in raspbian now according to the docs, so i'm guessing most people will only have issues if they are running something other than raspbian. that pull request will definitely help us notify people of the requirement.