adafruit/Adafruit-Pi-Finder #22: smbpasswd reports failure in occidentalis postinst

brennen, 2015-01-26:
```
Restarting Avahi mDNS/DNS-SD Daemon: avahi-daemon.
Restarting Netatalk Daemons (this will take a while)Stopping Netatalk Daemons: afpd cnid_metad papd timelord atalkd.
..Starting Netatalk services (this will take a while):  cnid_metad afpd.
done.
Mismatch - password unchanged.
Unable to get new password.
Stopping Samba daemons: nmbd smbd.
Starting Samba daemons: nmbd smbd.
```

toddtreece, 2015-01-26:
@brennen have you changed the default pass for the pi user?

brennen, 2015-01-26:
Nope - I've been keeping this install as stock as possible for testing bootstrap stuff.

That does maybe raise the question of whether we should mandate or strongly encourage the password change, but yeah, I guess this is really only a bug insofar as someone might notice it and be a little confused.  Probably not worth worrying about.

toddtreece, 2015-01-26:
I think the bug is here: https://github.com/adafruit/Adafruit-Pi-Finder/blob/master/packages/occidentalis/DEBIAN/postinst#L23

I'll test & fix

jakymiwm, 2015-05-05:
I just installed this on a fresh PI and I am not able to access the smb share. The credentials are bad.
FYI, my PC is on a domain. I made sure the username was raspberrypi\pi

I think it is in: https://github.com/adafruit/Adafruit-Occidentalis/blob/master/packages/occidentalis/DEBIAN/postinst#L32

jakymiwm, 2015-05-22:
I did a pull request on Occidentalis repo, not sure if anyone has seen it.
https://github.com/adafruit/Adafruit-Occidentalis/pull/4
Removing the '-ne' and '\n' seem to fix the "Mismatch - password unchanged." error.

toddtreece, 2015-05-24:
fixed by adafruit/Occidentalis#4

closed 2015-05-24 by toddtreece