adafruit/Adafruit-Pi-Finder #30: upload test

ladyada, 2015-02-01:
[image]

seems to work but umm maybe could use some ncurses? :)

ladyada, 2015-02-01:
shoot... :(
[image]

toddtreece, 2015-02-01:
@ladyada the [1.2-beta pre-release](https://github.com/adafruit/Adafruit-Pi-Finder/releases/tag/v1.2-beta) has new builds that should fix the upload percent display issue. How big was the file you sent?

ladyada, 2015-02-01:
lol i ran out of space, sorry i thought it was a 500K file but it was 500M. whee! go pats!

are you using scp or 'hand-transferring'?

toddtreece, 2015-02-01:
@ladyada i'm using a JS implementation of the SSH file transfer protocol. i can check to see if the version implemented allows for checking the free space available on the target pi

ladyada, 2015-02-01:
hmm i downloaded the pibootstrap 1.2 like an hour ago - should i grab it again to test the curses thing?

brennen, 2015-02-01:
> i can check to see if the version implemented allows for checking the free space available on the target pi

Maybe just an invocation of `df` on the target first?

toddtreece, 2015-02-01:
@brennen yeah. i was hoping to avoid parsing the output of `df`, but I guess I could do all of the prep in bash first and read the output.

@ladyada i replaced them right before commenting, but i don't think there's a reason to grab them just to test that. i'm pretty sure that bug is fixed.

brennen, 2015-02-01:
> but I guess I could do all of the prep in bash first and read the output.

Hmm, yeah. Maybe:

```
brennen@exuberance 16:19:15 /home/brennen $  df /home --output=avail | tail -1
128297852
```

toddtreece, 2015-02-01:
@brennen the installed version of `df` doesn't have `--output`. this is the best i could come up with:
```
df /home/pi | tail -1 | tr -s ' ' | cut -f 4 -d ' '
```

toddtreece, 2015-02-02:
this issue should be fixed in [v1.2.1-beta](https://github.com/adafruit/Adafruit-Pi-Finder/releases/tag/v1.2.1-beta)

ladyada, 2015-02-02:
all fixed & tested!

closed 2015-02-02 by ladyada