adafruit/Adafruit-Pi-Finder #36: RasPi 2 config changes

toddtreece, 2015-02-03:
via @ladyada:

> I am working on getting a BMP085 working. It would work on one SD card, but
not another. Turns out the problem is that the Raspberry Pi firmware was
more up to date one the SD card that didn't work than the SD card that
worked. Unfortunately with the release of RaPi 2, the Raspberrypi.org
website has been hard to get into. The following text must be added to the
/boot/config.txt file.
```
#to turn on I2C
dtparam=i2c1=on
#to turn on SPI
dtparam=spi=on
```

via @tdicola:
> also be careful if people modify /boot/config.txt from a windows pc with notepad it will screw the line endings. this happens to people on the BBB all the time and they think they bricked it since it won't boot

ladyada, 2015-02-03:
still investigating. plz hold before any edits :)

ladyada, 2015-02-03:
verified, please concat
    dtparam=i2c1=on
to /boot/config.txt
(thats all you gotta do for now)

toddtreece, 2015-02-04:
@ladyada: I have a question about something that was mentioned in this post: http://raspberrypi.stackexchange.com/a/27074

> I2C
Add dtparam=i2c1=on (or dtparam=i2c0=on on old models)

I'm wondering if we should be concerned about the *or dtparam=i2c0=on on old models* caveat, or should we just add `dtparam=i2c1=on` and call it good?

ladyada, 2015-02-04:
ahh the dtparam=i2c0=on line is for Pi B rev 1's - they are fairly rare. i'd stick with just
    dtparam=i2c1=on
and maybe put a # note  before saying
    # Turn on i2c (bus #1) for Raspberry Pi 2 and Pi 1 Model A, B (rev 2) and B+. If you have a Raspberry Pi Model B rev 1 with the i2c bus 0 instead, change this line to dtparam=i2c0=on

toddtreece, 2015-02-04:
@ladyada good call w/ the comment.

i think two of my test pis are rev 1's. i'll phase those out as soon as i get a test environment setup with net booting. having multiple is handy right now so i don't have to wipe a SD every time i want to test a fresh image against the bootstrap.

ladyada, 2015-02-05:
apparently there's another line you have to add!
    dtparam=i2c_arm=on
via https://forums.adafruit.com/viewtopic.php?f=10&t=67778&p=343544