adafruit/Adafruit-Pi-Finder #37: extra -e in /etc/modules

ladyada, 2015-02-03:
$ cat /etc/modules
# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.
# Parameters can be specified after the module name.

snd-bcm2835
-e i2c-bcm2708
i2c-dev
===============
there shouldn't be a -e before i2c-bcm2708 - i uninstalled and reinstalled occi and verified its still there

brennen, 2015-02-03:
I think this is coming in [via the occidentalis postinst](https://github.com/adafruit/Adafruit-Pi-Finder/blob/master/packages/occidentalis/DEBIAN/postinst#L15).  Checking why the `-e` would be treated as a literal and not a flag to `echo`.

brennen, 2015-02-03:
Ah, ok.  We're using `/bin/sh` here, which has an echo that doesn't offer `-e`. Sorta-fixed with c9271c7c, will see about papering over error on upgrade.

toddtreece, 2015-03-30:
this one should be fixed, and the code is no longer housed in this repo. closing

closed 2015-03-30 by toddtreece