adafruit/Adafruit-Pi-Finder #38: Raspberry Pi A+

tomasmcm, 2015-02-06:
Since the Raspberry Pi A+ doesn't have ethernet port I'm using a usb to ethernet adapter to connect to it, I can ssh to the pi via the terminal, but the Adafruit-Pi-Finder can't find it.
Is there a way to manually insert the IP address and jump to the Bootstrap section?

toddtreece, 2015-02-06:
@tomasmcm you can run this via ssh as the `pi` user to start the bootstrap:
```
curl -SLs https://apt.adafruit.com/install | sudo bash
```

The finder allows you to pass a couple config options, but you can set them from the Pi using the info found in [the occi section of the readme](https://github.com/adafruit/Adafruit-Pi-Finder#occidentalistxt). Let us know if that doesn't work for you.

We are working on a way to find Pi's over WiFi and non-standard ethernet adapters, and hopefully that should be out soon.

tomasmcm, 2015-02-06:
Yes, that's what I've just did :D

toddtreece, 2015-02-06:
Great! I'm going to close this one, but let us know if you have any more issues.

closed 2015-02-06 by toddtreece