adafruit/Adafruit-Pi-Finder #39: Allow more advanced network setups

gpgekko, 2015-02-07:
I noticed the finder code only looks in the last part of the user's IP address. However, in my home network (and probably quite a few business networks), the full 192.168.0.0/16 block is used. I'd like a configuration parameter to work with this. It would probably default to the current behavior for performance reasons, but either allowing a larger search or allowing to specify the search range manually would be nice.

toddtreece, 2015-02-07:
@gpgekko we might try to switch to something like nmap instead of sweeping through IPs with ping, but experimenting with bundling nmap for all platforms might take a couple weeks. Once that happens, I'll show a range config option after the first failed search.

ladyada, 2015-02-08:
@toddtreece perhaps we can look at the netmask to guess what subnet to search for? 99% of computers have  255.255.255.0 netmask -> 256 IP's to search. bigger networks would have a 255.255.0.0 -> 256 * 256 IPs (altho usally they have something more like 255.255.192.0)
http://www.sput.nl/internet/netmask-table.html

toddtreece, 2015-02-08:
yeah, that's a good idea. i was thinking of searching the first set of 256, and doing something like that on the second round. we could present the guessed search range to the user so they could modify it before searching again. searching 256 * 256 would take a long time, and i was thinking doing that automatically could cause issues at places like universities where multiple people might have Pi's plugged in with the default password.

ottoboni, 2015-02-09:
Maybe also just add a input field in the start view where you can fill in the ip for the raspberry pi. 
And yes I know it's a finder but sometimes it can't find it and it's frustrating to know that you have the IP.

gingerman, 2015-03-01:
Hello, 
I'd love to be able to have the functionality to create an ad-hoc wireless network from the Pi, so that I could push commands from my iPhone/iPad to my Pi while running it on a powerpack out and about.
( So that I can get my light painting project fully portable ). Great work so far - thanks.