adafruit/Adafruit-Pi-Finder #43: Add support to run CCHS Internet of Things setup

alecthegeek, 2015-02-09:
I'd like to add a feature so that users can run my IoT setup scripts.
(See  https://github.com/alecthegeek/CCHS_Raspian_for_IoT)

it would be something along the lines of `-z 1` and `-z 2` (it's a 2 phase setup)

Would you accept that sort of change or should I keep it in my own version?

toddtreece, 2015-02-10:
@alecthegeek the shell script has been moved to https://github.com/adafruit/Adafruit-Occidentalis, so I'm going to close this one out. it looks like your repo has some really cool stuff in it, but i think it might be out of the scope of the project to support it as an option directly since there is a lot of overlap with the occidentalis debian package.

if there are some changes you would like to see in occidentalis, feel free to file issues there and we can take a look.

closed 2015-02-10 by toddtreece

alecthegeek, 2015-02-10:
Sorry -- I didn't express myself well. I meant add an option to finder.sh to run the IoT Scripts instead of the Occidentals setup if wanted.

However since opening the issue, I've looked at Occidentals 0.3 and I think there are some great opportunities to improve both.

1. occi: I have attempted something similar in my project, however on first read your application looks more mature (using Perl was good idea there). There are however a couple of features that I think are worth adding
    a. support for a host name based on the CPU serial number in case you have a room full of Pis
    b. ability to bring the Pi up as Wifi access point. (I'm still working on this)

2. I have a lot more software and library installers (node, nodebots, golang etc). Some of which is not just an apt-get away :-(. However some of the installs are a bit hacky (clone a git repo and run a build/install script).

I'll look at the Adafruit-Occidentalis repo over the next few and see what ideas I can come up with.

Thanks

Alec