adafruit/Adafruit-Pi-Finder #45: SSH Times Out, Fails Handshake

ssharpjr, 2015-03-04:
RPI Finder finds my Pi just fine.  When I leave all the defaults, SSH throws a timed out error.  This is a fresh install of 2015-02-16-raspbian-wheezy.  I can SSH into it fine with putty.  I'm using Windows 8.1 Pro.

toddtreece, 2015-05-08:
i just updated the finder to use the latest version of [ssh2](https://github.com/mscdex/ssh2). you should get the update the next time you launch the pi finder if you are using v2.0+. if you are using v1.x, you can grab the latest release here: https://github.com/adafruit/Adafruit-Pi-Finder/releases

toddtreece, 2015-07-09:
@ssharpjr i'm going to close this one. feel free to reopen if it's still an issue after updating.

closed 2015-07-09 by toddtreece