adafruit/Adafruit-Pi-Finder #48: Can't find Pi with Ethernet sharing on Mac

joshmarinacci, 2015-04-02:
I'd like to set up my Pi but I don't have ethernet. (Not actually true, but let's assume so for the sake of this bug).  I plug my Pi into my Mac with ethernet and turn on Internet Sharing. The Mac connects to the Internet using wifi.  PiFinder can't find my pi, even though I can login to the pi directly and see with the console that it has an IP address.

Looking through the PiFinder code I think I understand the problem.  PiFinder scans 0-255 on the subnet of the main network interface. On my mac this is 192.168.1.x.  Internet Sharing creates *another* interface at 192.168.2.x.  If I scan that subnet it can find it.

So I think PiFinder should be updated to either:

* let the user choose a subnet to scan through
or, better
* detect *all* of the interfaces on the computer and scan them all.

On Mac, you can get the list of interfaces with ifconfig -a | grep 'broadcast'. This returns two lines with valid IP addresses and skips all of the various dongles and inactive devices.

If this is something you'd like I think I could write a patch for it.

toddtreece, 2015-04-03:
@joshmarinacci contributions are definitely welcome, and i like that idea of grabbing all of the interfaces. i have been trying to find a better way to make searching fast and configurable. this is a great start and should handle a lot of cases.

joshmarinacci, 2015-04-16:
I'm starting to work on this. Trying to build on mac. Why do I need Xquartz and wine for an Atomshell app?

Looks like i go to main, npm install, grunt build --force,  then cd ../build/darwin/atom-shell, then open ./Pi\ Bootstrap.app.  Right?

joshmarinacci, 2015-04-16:
The build system is a bit confusing it me. Assuming I understand the scripts correctly,  the main/ package launches atom shell with a stub that npm installs adafruit-pi-finder, which is the actual app.  However, this install always comes from the npm public repo, so changing code in package/ won't do anything until it's published. Is that right?  I've hacked it with a symlink for development, but it would be handy to have a test launcher during dev.

so, it looks like I can get all of the info i need from os.networkInterfaces. I'll make a patch which goes through all subnets, not just the default one.

toddtreece, 2015-04-17:
@joshmarinacci wine is needed for changing a few things in the Windows build. It's a bit confusing because the `main` folder builds downloadable atom-shell app, and the `package` folder is the actual Pi Finder app that I push up to npm. Decoupling the finder from the atom-shell app allows me to access npm programmatically from the atom-shell app and automatically pull in updates to the Pi Finder from npm without requiring users to download new releases.

I test changes to the finder package by running `npm install -g atom-shell` and then running `atom-shell test.js` from the package folder. Builds from the `main` folder are only required for changes to files in that folder, everything in the `package` folder is released with `npm publish`.

I'll update the folder structure and add updated README info for building.  Taking a look at the pull req now!

toddtreece, 2015-04-17:
@joshmarinacci published to npm and tagged as stable. the auto update should pick up your changes now.  thanks again for the help!

commit e4f2c402 (2015-04-17): bumps pi finder package to 2.2.0

fixes #48
closed 2015-04-17 by toddtreece