adafruit/Adafruit-Pi-Finder #50: Pi Finder auto-update failed - ERROR

SalvorinFex, 2015-04-05:
I keep getting this error after my win 7 computer is logged in after about an hour and I have to reboot for the program to run correctly again.  I do not think the the auto-update should prevent the program from running completely, throw an error , sure, but it should still run with the non-updated version.

https://www.dropbox.com/s/18wujau81tly88f/Screenshot%202015-04-04%2020.06.18.png?dl=0

toddtreece, 2015-04-05:
@SalvorinFex i can change error, but it should only run once when the app is launched. it shouldn't be trying to update once it is running. i'll check into that.

SalvorinFex, 2015-04-05:
There are timers that the Pi Finder window's size gets messed up and I need top close it and try to open it again, that's when I run into this issue.

toddtreece, 2015-04-05:
this was fixed with the release of [2.01-beta](https://github.com/adafruit/Adafruit-Pi-Finder/releases/tag/v2.0.1-beta). please reopen this if you still have the issue after upgrading.

closed 2015-04-05 by toddtreece