adafruit/Adafruit-Pi-Finder #51: Pi Finder auto update failed! Are you connected to the internet? on Mac 10.10.2

ukmoose, 2015-04-09:
Having successfully used the app in the past, I downloaded a new version and dragged into my Applications folder.  On double clicking, the app launches and displays the following message "Pi Finder auto update failed! Are you connected to the internet?"  Clicking "OK" closes that window but nothing else appears.  

Using LittleSnitch the only network connection from the app appears to be an https connection to api.npmjs.org which is allowed and the connection is successful.  Even with all network traffic allowed to & from the app the same error occurs.  Any ideas?

toddtreece, 2015-04-09:
@ukmoose do you have any sort of firewall turned on? if so, could you stop it and try again? i am running 10.10.2 with the latest release and the app launches successfully. let me know if that isn't the issue.

ukmoose, 2015-04-09:
Only the Mac OS X firewall and Little Snitch,  with both disabled  i.e. OFF I still get the same error.

Last time I tried I did get the following error displayed on clicking OK in the Are you connected one.

Uncaught Exception:
Error: read ECONNRESET
    at exports._errnoException (util.js:734:11)
    at TLSWrap.onread (net.js:538:26)

toddtreece, 2015-04-09:
were you using 2.0 before, or an earlier release? 2.0 introduced the update feature, and 2.0.1 made it a bit more stable by falling back to the currently installed version. if you were using 2.0 before, can you try to re-download 2.0.1 and try with that?

@jwcooper just confirmed that it works for him as well, so it seems to be something with your specific network setup. are you behind a filter or firewall at a school or business?

ukmoose, 2015-04-11:
I was using release 1 before hence the reason I upgraded.
So I'm at home,  and on a second mac on the same network the app works as expected.

I'm using nodejs on the problem laptop so am confident that I can get a connection to npmjs.org

ukmoose, 2015-04-14:
Last night I fired up wireshark on the problem laptop and captured the network traffic.

[image]

As you can see from the image there's definitely a working network connection from the laptop to registry.npmjs.org which appears to be the only site that the app polls.

I've deleted the app, and the files its stores in
/Users/<username>/Library/Saved Application State/com.github.atom-shell.savedState/
Reinstalled and I still have the same problem.

ukmoose, 2015-04-14:
solved it!  
I ran "npm i adafruit-pi-finder" from the command line
and got the following error message:

npm ERR! Error: EACCES, mkdir '/Users/msetrem/.npm/asn1/0.2.1'
npm ERR!  { [Error: EACCES, mkdir '/Users/msetrem/.npm/asn1/0.2.1']
npm ERR!   errno: 3,
npm ERR!   code: 'EACCES',
npm ERR!   path: '/Users/msetrem/.npm/asn1/0.2.1',
npm ERR!   parent: 'ssh2' }
npm ERR! 
npm ERR! Please try running this command again as root/Administrator.

ran the command again with sudo.  and with that dependency installed the application now opens.

I guess the error message given by the app is more than slightly confusing!

toddtreece, 2015-04-14:
@ukmoose that's very strange. you shouldn't need root to access your own user's '.npm' folder. do you use node outside of the pi finder? maybe something happened with permissions before the install?

try running this in the terminal:
```
todd:~ todd$ cd ~
todd:~ todd$ ls -la | grep npm$
drwxr-xr-x  924 todd  staff  31416 Apr  9 11:05 .npm
```

is root the owner?

ukmoose, 2015-04-14:
Yes I use nodejs outside of pi-finder.
No .npm is owned by me, but some of the directories below .npm are owned by root which included the preexisting ans1 directory.

Either way, whilst Pi_finder is very useful, the error message is less than helpful

toddtreece, 2015-04-14:
Something still seems weird with the `.npm` folder permissions. My `.npm` folders on multiple machines have 100s of items, and they are all owned by the user and not by root. Sorry that took so long to debug! I'll see if I can find a clear way to provide more details when the update fails.

toddtreece, 2015-05-08:
@ukmoose I think I figured out a solution to avoid the npm cache perm issues. This should hopefully fix the issue: https://github.com/adafruit/Adafruit-Pi-Finder/releases/tag/3.0.0-beta

Here's the commit for the fix: 7bfeabc4f1ddf2b29d9acdd05df39c6c21c18302

toddtreece, 2015-05-22:
i'm going to close this since i haven't heard back since the release of 3.0. please reopen if it's still a problem after upgrading.

closed 2015-05-22 by toddtreece

jezmck, 2015-08-18:
I'm having this problem now. `v0.22.3` according to the version file, downloaded within the last week.

On Windows 7.

toddtreece, 2015-08-21:
@jezmck give the latest release a try

jezmck, 2015-08-21:
I've found it to work without updating, computer has restarted in the meantime.

prwiley, 2017-02-13:
I am having this same issue. On MacOS 10.12.2. I have the latest macOS version from github.  My internet connection is via WiFi. No idea what to do. Anyone have suggestions?

davidbell, 2017-02-24:
Just got Pi Finder and encountered the same problem, then I saw the comment here and ran it as Administrator and then it opened fine. (Win 10 64bit)

40thiefs, 2017-03-03:
i was having same problem but once i moved the program from download folder to application folder, it worked w/o any error message.

haynesherway, 2018-02-11:
I also had to move PiFinder from downloads to applications to get it to work.

alekbless, 2020-11-20:
Had to move it from download folder as well, could be helpful to add it to the docs.