adafruit/Adafruit-Pi-Finder #52: A JavaScript error occured in the browser process

goliatone, 2015-04-10:
The app throws a bunch of errors after clicking the **Find my Pi** button. I have to force quit to exit.

```
Uncaught Exception:
TypeError: Cannot read property 'on' of undefined
    at mac (/Users/goliatone/Downloads/pibootstrap_mac/Pi Bootstrap.app/Contents/Resources/app/node_modules/adafruit-pi-finder/arp.js:86:13)
    at arp (/Users/goliatone/Downloads/pibootstrap_mac/Pi Bootstrap.app/Contents/Resources/app/node_modules/adafruit-pi-finder/arp.js:15:5)
    at Finder.proto.arp (/Users/goliatone/Downloads/pibootstrap_mac/Pi Bootstrap.app/Contents/Resources/app/node_modules/adafruit-pi-finder/finder.js:70:3)
    at Finder.<anonymous> (/Users/goliatone/Downloads/pibootstrap_mac/Pi Bootstrap.app/Contents/Resources/app/node_modules/adafruit-pi-finder/finder.js:62:10)
    at ChildProcess.<anonymous> (/Users/goliatone/Downloads/pibootstrap_mac/Pi Bootstrap.app/Contents/Resources/app/node_modules/adafruit-pi-finder/ping.js:17:5)
    at emitTwo (events.js:87:13)
    at ChildProcess.emit (events.js:169:7)
    at Process.ChildProcess._handle.onexit (child_process.js:1057:12
```

```
Uncaught Exception:
Error: spawn arp EMFILE
    at exports._errnoException (util.js:734:11)
    at Process.ChildProcess._handle.onexit (child_process.js:1035:32)
    at child_process.js:1127:20
    at process._tickCallback (node.js:357:13)
    at Object.module.exports.showErrorBox (/Users/goliatone/Downloads/pibootstrap_mac/Pi Bootstrap.app/Contents/Resources/atom.asar/browser/api/lib/dialog.js:143:35)
    at process.<anonymous> (/Users/goliatone/Downloads/pibootstrap_mac/Pi Bootstrap.app/Contents/Resources/atom.asar/browser/lib/init.js:45:30)
    at emitOne (events.js:77:13)
    at process.emit (events.js:166:7)
    at process._fatalException (node.js:235:26)
    at ChildProcess.<anonymous> (/Users/goliatone/Downloads/pibootstrap_mac/Pi Bootstrap.app/Contents/Resources/app/node_modules/adafruit-pi-finder/ping.js:17:5)
```

```
Uncaught Exception:
Error: spawn arp EMFILE
    at exports._errnoException (util.js:734:11)
    at Process.ChildProcess._handle.onexit (child_process.js:1035:32)
    at child_process.js:1127:20
    at process._tickCallback (node.js:357:13)
```

```
  Model Name: 	MacBook Pro
  Model Identifier:	MacBookPro11,3
  Processor Name:	Intel Core i7
  Processor Speed:	2.3 GHz
  Number of Processors:	1
  Total Number of Cores:	4
  L2 Cache (per Core):	256 KB
  L3 Cache:	6 MB
  Memory:	16 GB
```

toddtreece, 2015-04-10:
thanks for the detailed bug report!

this should be fixed. if you are using version 2.0 or greater of the pi finder it will auto update on restart of the app. if you are using a pre-2.0 release, you should update to the latest release.

as a quick sanity check, could you paste in the output of running `ulimit -a` in the terminal?

closed 2015-04-10 by toddtreece

goliatone, 2015-04-10:
Thanks for the quick fix. Will try again. I ended up doing the bootstraping manually, thou.

```
core file size          (blocks, -c) 0
data seg size           (kbytes, -d) unlimited
file size               (blocks, -f) unlimited
max locked memory       (kbytes, -l) unlimited
max memory size         (kbytes, -m) unlimited
open files                      (-n) 256
pipe size            (512 bytes, -p) 1
stack size              (kbytes, -s) 8192
cpu time               (seconds, -t) unlimited
max user processes              (-u) 709
virtual memory          (kbytes, -v) unlimited
```

toddtreece, 2015-04-10:
@goliatone ah. ok. sorry about that. it was bumping up against the max open file limit because it sweeped all 256 addresses in parallel in the previous version. i added a parallel execution limit that should fix the problem in the future.

goliatone, 2015-04-10:
@toddtreece perfect! Terrific job, thank you!!

Marcus-ramassi, 2023-04-11:
[image]
erro ao instalar no w11

commit 0a0bade2 (2015-04-10): adds parallel execution limit

fixes #52
also adds better error handling for spawned tasks