adafruit/Adafruit-Pi-Finder #54: Uncaught Exception: Error: Attempt to unlock adafruit-pi-finder which hasn't been locked

devanp92, 2015-04-26:
On OSX. Opened the application and got this error (even with right-click + open): 

    A JavaScript error occured in the browser process
          Uncaught Exception:
      Error: Attempt to unlock /Applications/Pi Bootstrap.app/Contents/Resources/app/node_modules/adafruit-pi-finder, which hasn't been locked
          at unlock (/Applications/Pi Bootstrap.app/Contents/Resources/app/node_modules/npm/lib/utils/locker.js:66:11)
          at cb (/Applications/Pi Bootstrap.app/Contents/Resources/app/node_modules/npm/lib/install.js:1004:5)
          at /Applications/Pi Bootstrap.app/Contents/Resources/app/node_modules/npm/lib/install.js:1008:20
          at /Applications/Pi Bootstrap.app/Contents/Resources/app/node_modules/npm/lib/utils/locker.js:40:9
          at cb (/Applications/Pi Bootstrap.app/Contents/Resources/app/node_modules/npm/node_modules/lockfile/lockfile.js:149:38)
          at /Applications/Pi Bootstrap.app/Contents/Resources/app/node_modules/npm/node_modules/lockfile/lockfile.js:177:38
          at FSReqWrap.oncomplete (fs.js:77:15)

It has the proper read/write/execute permissions. Has anyone experienced this before?

jmsaavedra, 2015-04-30:
Identical exception for me on launch of .app

* OSX 10.10.3. 

```
A JavaScript error occured in the browser process
   Uncaught Exception:
Error: Attempt to unlock /Users/jmsaavedra/Apps/pibootstrap_mac/Pi
... 
```

toddtreece, 2015-04-30:
@devanp92 & @jmsaavedra sorry. i missed the original bug report. are either of you doing node.js development on the same machine?

jmsaavedra, 2015-04-30:
thanks for getting back! 

indeed, tons of node.js dev on this machine. no apps running when i try executing, however.

toddtreece, 2015-04-30:
@jmsaavedra i'm guessing there is something up with npm permissions. have you installed packages using `sudo npm install` before?

check out this issue: https://github.com/npm/npm/issues/4815

toddtreece, 2015-05-08:
@devanp92 & @jmsaavedra: I think I figured out a solution to avoid the npm cache perm issues. This should hopefully fix the issue: https://github.com/adafruit/Adafruit-Pi-Finder/releases/tag/3.0.0-beta

Here's the commit for the fix: 7bfeabc4f1ddf2b29d9acdd05df39c6c21c18302

toddtreece, 2015-05-22:
i'm going to close this one since i haven't heard back in a couple weeks. feel free to reopen if v3.0+ doesn't fix the issue.

closed 2015-05-22 by toddtreece