adafruit/Adafruit-Pi-Finder #56: Pi Finder can't find Pi

dlastro, 2015-08-24:
After downloading PiFinder to my iMac(Yosemite) I was able to connect to my RaspPi Model A the first time by WiFi (even without first connecting by ethernet. However, bootstrap via CLI failed to work.)

Upon attempting to connect to a RaspPi 2, running the latest Raspian (NOOBS v_1_4_1), failed with both Pi-Finder and Bootstrap. I have the RP2 connected to the Mac via direct Ethernet cable (have tried 3 so far). The RP network connection icon shows: "eth0: Configured 169.254.157.27/16." However Pi-Finder says "Your Pi could not be found".

Shut down RP2 and booted RP-A and launched Pi-Finder. Both terminal and bootstrap work with my WiFi network (Apple Airport Extreme). Bootstrap via CLI still failed to find RP. Nothing happens after "Username is 'Pi' "

Am I missing something with the Ethernet connection setup, or does Pi-Finder not work with RP2?

tdicola, 2015-08-27:
If you're connecting the Pi directly to a laptop are you using a crossover ethernet cable to make sure each device can talk to the other without an intermediate hub/router?  Unfortunately a direct connection with a normal ethernet cable won't work, but you could plug both devices into a hub/router and it should work.

craigreilly, 2016-03-23:
I have similar issue - PiFinder (Mac) can not find Pi via WiFi.
If Pi is plugged in to Ethernet - it is found - with 2 IP's - WiFi and Hardline.
At that point, I can unplug the hard line and WiFi will work.
I am heading to a convention with WiFi only. How can I ensure I can connect via WiFi?

ladyada, 2016-03-23:
yah its only ethernet at this time - it detects the MAC address!

craigreilly, 2016-03-23:
strange that it works on WiFi after it finds the unit. Shoot. I guess I'll have to bring a Router with me and setup with a small LAN to setup the WiFi.

toddtreece, 2016-03-23:
@craigreilly you can probably just use the terminal to find your pi since mdns/bonjour is enabled by default in the latest version of raspbian.

give this a shot:
```
ssh pi@raspberrypi.local
```

craigreilly, 2016-03-23:
thanks. what would happen if multiple units are connected to the network?
i only have 1 right now for testing.

toddtreece, 2016-03-23:
@craigreilly you would need to give each pi a unique hostname. i'd recommend posting your question over at the [adafruit forums](https://forums.adafruit.com/) if you need help getting that going.

craigreilly, 2016-03-23:
appreciate all the help. thank you.