adafruit/Adafruit-Pi-Finder #59: Build for Long-term Use

TheKitty, 2016-03-28:
Combine the two screens, put the Find My Pi button atop the main screen, hide the bottom unless connected.

Save session info from one run to the next.  Start the ip scanning near/on the last IP address to possibly speed things up.

If a pi is already bootstrapped, it doesn't need it again, yes?

toddtreece, 2016-03-29:
@TheKitty the pi finder's UI has needed an update from pretty much day 1, but I have been wondering if this tool is necessary anymore since the latest releases of raspbian include mdns/bonjour support. user's should be able to easily find their pi's using the hostname. what do you think?

TheKitty, 2016-03-30:
Still useful would be my vote.

SalvorinFex, 2016-06-02:
I still use use it and plan on continuing to use it.
An Update would be very welcome.

Nsrose, 2016-06-08:
Not only do I plan on continuing to use it, but I also plan on directing all my customers (to whom I sell modded Pis) to use it to install their wifi credentials. I'm in high favor of these updates.