adafruit/Adafruit-Pi-Finder #61: Pi Finder auto update fail. Are you connected to the internet? (Yes, I am)

NauchtanRobotics, 2016-04-11:
Why is running Pi Finder conditional on connecting to the internet to update? Surely it should be able to run on my Windows 8.1 PC as released?  Plus, I was connected to the internet.  
[image]

joanaz, 2017-01-12:
I'm having the same problem with the 3.0.0 release. I'm using the school wifi, so there might be a firewall, any way to get around it?

Antiparras, 2017-01-14:
Start the app as Administrator and it will go straight in :)

moronidiot, 2018-03-07:
Im having this issues too, Even with adminsitrator permission.

macnaughtond, 2018-03-07:
I am using Linux now and I just ping all devices on my home network.  The command will be similar if you are using a Windows Power-Shell Terminal. From experience, all ip addresses on my home network start with 192.168.43...   So I make a loop to cycle the last digit from 192.168.43.1 through to 192.168.43.255 like this:
for ip in $(seq 1 254); do         ping -c 1 192.168.43.$ip | grep "bytes from" | cut -d " " -f 4 | cut -d ":" -f 1 &     done
Do this once with your Pi turned off, and once with it turned on, and note the difference. Make sure your home router/hot-spot is turned on. 
The you can simply type:
ssh pi@192.168.##.###     
type 'yes' to add the address permanently, and enter raspberry as the password.