adafruit/Adafruit-Pi-Finder #63: Pi Finder UI won't load on OSX El Capitan

cheshrkat, 2016-06-05:
System: mid-2011 macbook air, OSX El Capitan 10.11.5

Steps taken:

- Downloaded 3.0.0 OSX package
- extracted with default archive util
- dragged into applications
- right-click-open
- accepted the prompt to open unknown app.

Expected: pi finder window to show.

Actual: Pi Finder icon shows in doc and header, but nothing else shows. Can be opened and quit, no error messages, but no UI shows (can't use the finder tool).

sanderdb01, 2016-11-09:
im getting the same error

alexking, 2016-12-13:
This is happening to me on El Capitan as well. Looking into it a little, it appears [this section is failing](https://github.com/adafruit/Adafruit-Pi-Finder/blob/master/wrapper/src/main.js#L12-L22) 

If you edit the code inside `PiFinder.app` (via Show Package Contents) to print the `err` field, you can see this - 

`Error: EROFS: read-only file system, mkdir '/private/var/folders/[removed]/d/PiFinder.app/Contents/Resources/app/npm_cache/adafruit-pi-finder/2.4.1/package'`

Which leads to the `try` on the next line giving `Error: Cannot find module 'adafruit-pi-finder'`. 

So there's some sort of [npm](https://www.npmjs.com) issue where it can't write inside of the application package. I should note that I have copied it to my drive, it's not sitting inside a disk image. The odd thing is I'm able to write inside of the app package. Supposedly this [might have to do with El Capitan's system integrity protections](https://github.com/electron/electron/issues/3151).

As a quick workaround, if you just want to find your Pi and have npm/node already installed on your system, you can actually manually install the package into the application. Just do - 

`cd ~/mydir/PiFinder.app/Contents/Resources/app
npm install adafruit-pi-finder@stable`

After running that, I was able to restart the app and use it no problem.

hiroMTB, 2017-09-06:
I've got same problem but this still works in 2017. thanks!