adafruit/Adafruit-Pi-Finder #75: OS X: will not run from Downloads or Desktop folder

brianhanifin, 2017-10-16:
The app has to be moved to the Applications folder before it can run successfully. It would nice if I could just run it from the Downloads folder so I could delete it after I am done running it.

**Steps to reproduce**
1. Download and extract _PiFinder-3.0.0-osx-x64.zip_.
2. Run PiFinder.app from the Downloads folder.

**Expected Results**
Pi Finder opens and displays the IP Addresses of the Raspberry Pi devices on my network.

**Actual Results**
* Dialog box appears:
```
ERROR
Pi Finder auto update failed! Are you connected to the internet?
```
* Clicking the "OK" button results in the dialog box closing.
* PiFinder appears to be running (as indicated by the "Adafruit Pi Finder" menu at the top of the screen.
* However no new window appears, and the only option in that menu is "Quit".
* Right clicking on the PiFinder icon in the dock and choosing "Show all windows" yields a message on the desktop stating "No available windows".

**Solution**
* Move PiFinder.app to the Applications folder before running.

**My system**
Mac (Retina 5K, 27-inch, 2017)
OS X 10.13 "High Sierra"

sidpagariya, 2017-12-04:
Thanks! That fixed the issue! Maybe we should see why it wouldn't work without the move to /Applications...

Thanks,
Sid.

dj2mn, 2020-01-14:
FWIW I just got the same error on Catalina 10.15.2 .. but moving it to /Applications hasn't helped much apart from getting rid of the "are you connected to the internet" message, now it launches but the app window doesn't appear. To be honest I'm not sure whether it'd work anyway due to the funky network restrictions we have here at the office, but it'd be nice to at least see a window that says it can't find a PI..

dvncan, 2021-06-24:
Thanks this worked on BigSur, I was trying everything to get rid of this error: 

```Pi Finder auto update failed! Are you connected to the internet?```

Moving the application to my applications folder fixed it!