adafruit/Adafruit-Pi-Finder #78: Pi Finder Can't Find My Pi

tedog123456789, 2020-01-03:
I have my Pi all set up and connected to ethernet but when I click the "Find My Pi" button it fails and gives the error: "Your Pi could not be found. Are you sure it is attached to the network?" I'm far from an expert at this so it's probably something I did wrong. 

OSX
Raspberry Pi 4
Pi Finder 3.0.0

guigro, 2020-02-17:
Same here ! Didn't tried with a rpi 3 to see if the issue is with new Pi.

adiazulay, 2020-06-17:
I have the same issue with my Pi 4. I can ping and ssh into the pi just fine from command line, but pi finder doesn't find it.

Windows 10
Raspberry Pi 4 4GB
Pi Finder 3.0.0

chetmcmasterson, 2021-01-13:
Confirming this issue persists. Pi Finder can't find my Pi 4s. They're on the same wired ethernet as the Pi Finder machine, and reply to ping and ssh.

This looks like a great tool for the bootstrapping capability--here's hoping it isn't dead!

ladyada, 2021-01-13:
hi this is not maintained. if someone wants to submit PRs to fix we'll take a look. ideally someone would take it over.

dvncan, 2021-06-24:
Same issue for me, however could this because I have renamed the hostname to a custom one? Is the device hostname hardcoded into the source?